WebPlatformForEmbedded/meta-wpe #330: wpeframework-3.0+gitAUTOINC+367ca8bb39-r0 compilation error

VarshithaRani, 2019-07-30:
ERROR: wpeframework-3.0+gitAUTOINC+367ca8bb39-r0 do_compile: Function failed: do_compile
...
 | DEBUG: Executing shell function do_compile
| NOTE:VERBOSE=1 cmake --build /home/varshitha/yocto/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpeframework/3.0+gitAUTOINC+367ca8bb39-r0/build --target all -- -j 4
........
 /home/varshitha/yocto/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpeframework/3.0+gitAUTOINC+367ca8bb39-r0/git/Source/compositorclient/Wayland/Implementation.h:119:65: error: invalid static_cast from type 'wl_egl_window* const' to type 'EGLNativeWindowType' {aka 'long unsigned int'}
|   119 |                 return (static_cast<EGLNativeWindowType>(_native));
|       |                                                                 ^
| In file included from /home/varshitha/yocto/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpeframework/3.0+gitAUTOINC+367ca8bb39-r0/git/Source/compositorclient/Wayland/Westeros.cpp:13:
| /home/varshitha/yocto/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpeframework/3.0+gitAUTOINC+367ca8bb39-r0/git/Source/compositorclient/Wayland/Implementation.h:30:8: note: class type 'wl_egl_window' is incomplete
|    30 | struct wl_egl_window;
|       |        ^~~~~~~~~~~~~
| /home/varshitha/yocto/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpeframework/3.0+gitAUTOINC+367ca8bb39-r0/git/Source/compositorclient/Wayland/Westeros.cpp: In member function 'bool WPEFramework::Wayland::Display::SurfaceImplementation::Connect(void* const&)':
| /home/varshitha/yocto/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpeframework/3.0+gitAUTOINC+367ca8bb39-r0/git/Source/compositorclient/Wayland/Westeros.cpp:653:65: error: invalid static_cast from type 'std::nullptr_t' to type 'EGLNativeWindowType' {aka 'long unsigned int'}
|   653 |                         static_cast<EGLNativeWindowType>(nullptr),
|       |

VarshithaRani, 2019-07-30:
I am getting this error when I am taking the recent build of meta-wpe layer from master branch.

wouterlucas, 2019-07-30:
Try removing vc4graphics from your machine features.

VarshithaRani, 2019-07-31:
Thanks @wouterlucas. It's working

closed 2019-07-31 by VarshithaRani

psivasubramanian, 2019-07-31:
This commit https://github.com/Khan3033/Thunder/commit/5728004273ab1ae3aaf7f868cff1e2dc603025b1 can   address this issue.

limbouser, 2020-05-27:
@psivasubramanian  The commit you mentioned is no longer available. Is their any way I can get it?