WebPlatformForEmbedded/meta-wpe #352: yocto build recipe "meta-eglfs-image" for raspberrypi3-64 fails as 'vulkan' needs 'wayland' or 'x11'

thadivaka-gopikrishna, 2019-09-06:
Nothing PROVIDES 'vulkan' (but /home/gopi/my-yocto-project/_toaster_clones/_git___git.yoctoproject.org_poky_warrior/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.14.4.bb DEPENDS on or otherwise requires it)
vulkan was skipped: one of 'wayland x11' needs to be in DISTRO_FEATURES

Required build target 'wpe-eglfs-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['wpe-eglfs-image', 'packagegroup-wpewebkit', 'wpebackend-rdk', 'wpeframework-plugins', 'wpewebkit', 'gstreamer1.0-plugins-bad', 'vulkan']
[local.conf.txt](https://github.com/WebPlatformForEmbedded/meta-wpe/files/3583544/local.conf.txt)
[bblayers.conf.txt](https://github.com/WebPlatformForEmbedded/meta-wpe/files/3583546/bblayers.conf.txt)

d21d3q, 2019-09-18:
I've alsso encountered similar problem but removing `vulkan` via `DISTRO_FEATURES_remove` made it work. so 

More about that issue, I tried to build `wpe-eglfs-image` with poky (not yoe as suggested here) and first of all bitbake was complaining about `wayland`, after removing, it complained about `x11`, and after removing one, about `vulkan`

So with:
```
DISTRO_FEATURES_remove += " wayland x11 vulkan"
```
it seems to work(around).

kraj, 2019-09-18:
Add vulkan to conflict features in the image so we can get a decent error message

d21d3q, 2019-09-18:
But `vulkan` does not cause direct conflict. It just [requires `wayland` or `x11`](https://github.com/lgirdk/poky/blob/330bd591ec6a7badbff0b1e7e7d7dfdd248a9afb/meta/recipes-graphics/vulkan/vulkan-tools_1.1.108.bb#L14) inside, and that causes conflict.
I think that there is some elegant yocto-way of coping with such situations? (I am new to yocto world so I'm just guessing)

closed 2019-09-19 by thadivaka-gopikrishna