WebPlatformForEmbedded/meta-wpe #379: meta-wpe RPI build failure in bluetooth

vkanupuru, 2020-01-17:
Raspberry Pi build failed in bluetooth 

Error:
ERROR: wpeframework-plugins-3.0+gitrAUTOINC+a84313d035-r1 do_configure: Function failed: do_configure (log file is located at /mnt/home/cpokur207/wpealone-new/rpi-ml-build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework-plugins/3.0+gitrAUTOINC+a84313d035-r1/temp/log.do_configure.23739)
ERROR: Logfile of failure stored in: /mnt/home/cpokur207/wpealone-new/rpi-ml-build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework-plugins/3.0+gitrAUTOINC+a84313d035-r1/temp/log.do_configure.23739
Log data follows:
| DEBUG: Executing python function sysroot_cleansstate
| DEBUG: Python function sysroot_cleansstate finished
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is GNU 6.4.0
| -- The CXX compiler identification is GNU 6.4.0
| -- Check for working C compiler: /mnt/home/cpokur207/wpealone-new/rpi-ml-build/tmp-glibc/sysroots/x86_64-linux/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-gcc
| -- Check for working C compiler: /mnt/home/cpokur207/wpealone-new/rpi-ml-build/tmp-glibc/sysroots/x86_64-linux/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-gcc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Check for working CXX compiler: /mnt/home/cpokur207/wpealone-new/rpi-ml-build/tmp-glibc/sysroots/x86_64-linux/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-g++
| -- Check for working CXX compiler: /mnt/home/cpokur207/wpealone-new/rpi-ml-build/tmp-glibc/sysroots/x86_64-linux/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-g++ -- works
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| CMake Error at BluetoothControl/CMakeLists.txt:7 (find_package):
|   By not providing "FindWPEFrameworkBluetooth.cmake" in CMAKE_MODULE_PATH
|   this project has asked CMake to find a package configuration file provided
|   by "WPEFrameworkBluetooth", but CMake did not find one.
| 
|   Could not find a package configuration file provided by
|   "WPEFrameworkBluetooth" with any of the following names:
| 
|     WPEFrameworkBluetoothConfig.cmake
|     wpeframeworkbluetooth-config.cmake
| 
|   Add the installation prefix of "WPEFrameworkBluetooth" to CMAKE_PREFIX_PATH
|   or set "WPEFrameworkBluetooth_DIR" to a directory containing one of the
|   above files.  If "WPEFrameworkBluetooth" provides a separate development
|   package or SDK, be sure it has been installed.
| 
| 
| -- Configuring incomplete, errors occurred!
| See also "/mnt/home/cpokur207/wpealone-new/rpi-ml-build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework-plugins/3.0+gitrAUTOINC+a84313d035-r1/build/CMakeFiles/CMakeOutput.log".
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /mnt/home/cpokur207/wpealone-new/rpi-ml-build/tmp-glibc/work/cortexa7hf-neon-vfpv4-oe-linux-gnueabi/wpeframework-plugins/3.0+gitrAUTOINC+a84313d035-r1/temp/log.do_configure.23739)
ERROR: Task (/mnt/home/cpokur207/wpealone-new/meta-wpe/recipes-wpe/wpeframework/wpeframework-plugins_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3240 tasks of which 3239 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /mnt/home/cpokur207/wpealone-new/meta-wpe/recipes-wpe/wpeframework/wpeframework-plugins_git.bb:do_configure
Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
cpokur207@dvm-hoa-029:~/wpealone-new/rpi-ml-build$ vi conf/local.conf 
cpokur207@dvm-hoa-029:~/wpealone-new/rpi-ml-build$ 
cpokur207@dvm-hoa-029:~/wpealone-new/rpi-ml-build$

wouterlucas, 2020-02-05:
Is this on morty or master?

vkanupuru, 2020-02-05:
This was resolved in the latest sync and i am able to build the code from morty branch

wouterlucas, 2020-02-06:
Great! Closing this

closed 2020-02-06 by wouterlucas