WebPlatformForEmbedded/meta-wpe #390: bb expansionerror as expecting 'licenses' dir

moorthy-bs, 2020-02-05:
conf/layer.conf defines LICENSE_PATH and latest bitbake throws error as the expected license directory is not available

petegriffin, 2020-02-05:
ERROR: Unable to parse Var <AVAILABLE_LICENSES[:=]>
Traceback (most recent call last):
  File "Var <AVAILABLE_LICENSES[:=]>", line 1, in <module>
  File "/home/arthur/workspace/linaro/lhg/oe/imx8m-wpewebkit/build-rpb-wayland/conf/../../layers/openembedded-core/meta/classes/license.bbclass", line 275, in available_licenses(d=<bb.data_smart.DataSmart object at 0x7fd42147b5c0>):
         for lic_dir in lic_dirs:
    >        licenses += os.listdir(lic_dir)
     
bb.data_smart.ExpansionError: Failure expanding variable AVAILABLE_LICENSES[:=], expression was ${@' '.join(available_licenses(d))} which triggered exception NotADirectoryError: [Errno 20] Not a directory: '/home/arthur/workspace/linaro/lhg/oe/imx8m-wpewebkit/build-rpb-wayland/conf/../../layers/meta-wpe/licenses'

wouterlucas, 2020-02-05:
Makes sense as that doesnt exist. We should probably only point to the license file, is there another var for that?

moorthy-bs, 2020-02-06:
License file mentioned in LIC_FILES_CHKSUM can be initially referred from the path(s) mentioned in LICENSE_PATH. If the given file not available in any of the path mentioned in LICENSE_PATH, it then refers from the source ${S}.
I found a file 'LICENSE' in the meta-wpe base directory. I think we can move this to meta-wpe/licenses directory and same can be referred from any meta file (bb) or COREBASE path to be added to LICENSE_PATH. Make sense?
Ref. [https://www.yoctoproject.org/docs/3.1/ref-manual/ref-manual.html]
@kraj FYI

wouterlucas, 2020-02-06:
Sure, that's fine with me.

PR #393 (2020-02-11): [license-path]: LICENSE files to be in specific dir
In order to resolve bb Expansion Error when build with
latest oe-core, LICENSE file is moved to 'licenses' directory
as this is mentioned in LICENSE_PATH

Fix: #390
commit 30c8c211 (2020-02-11): [license-path]: LICENSE files to be in specific dir

In order to resolve bb Expansion Error when build with
latest oe-core, LICENSE file is moved to 'licenses' directory
as this is mentioned in LICENSE_PATH

Fix: #390
closed 2020-02-11 by wouterlucas