WebPlatformForEmbedded/meta-wpe #396: AVAILABLE_LICENSES error when compiling yoe, based on meta-wpe README instructions

fellowgeek, 2020-02-08:
```
bitbake wpe-eglfs-image
ERROR: Unable to start bitbake server (None)
ERROR: Server log for this session (/mnt/500GB/rpi-linux/yoe-distro/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 28194 at 2020-02-08 11:45:22.615768 ---
ERROR: Unable to parse Var <AVAILABLE_LICENSES[:=]>
Traceback (most recent call last):
  File "Var <AVAILABLE_LICENSES[:=]>", line 1, in <module>
  File "/mnt/500GB/rpi-linux/yoe-distro/sources/openembedded-core/meta/classes/license.bbclass", line 275, in available_licenses(d=<bb.data_smart.DataSmart object at 0x7fb6fc47b240>):
         for lic_dir in lic_dirs:
    >        licenses += os.listdir(lic_dir)
     
bb.data_smart.ExpansionError: Failure expanding variable AVAILABLE_LICENSES[:=], expression was ${@' '.join(available_licenses(d))} which triggered exception FileNotFoundError: [Errno 2] No such file or directory: '/mnt/500GB/rpi-linux/yoe-distro/sources/meta-wpe.git/licenses'
```

wouterlucas, 2020-02-11:
This should be fixed by https://github.com/WebPlatformForEmbedded/meta-wpe/pull/393

wouterlucas, 2020-02-11:
Re-open if it still happens with #393 merged in.

closed 2020-02-11 by wouterlucas