WebPlatformForEmbedded/meta-wpe #417: Can't compile because of missing stime

timemaster5, 2020-03-16:
Hello, I am trying to compile Yoe Distro with YoeDistro/meta-wpe layer, but I can't get over this issue:
```
Framework::Core::SystemInfo::SetTime(const WPEFramework::Core::Time&)':
| /home/ubuntu/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpeframework/3.0+gitAUTOINC+9629e13e39-r0/git/Source/core/SystemInfo.cpp:327:13: error: 'stime' was not declared in this scope; did you mean 'ctime'?
|   327 |         if (stime(&value) != 0) {
|       |             ^~~~~
|       |             ctime
```
I think this is because of compiler version, but I am using the recommended docker image _yoedistro/yoe-build:buster_ so I though I should be fine.

This is for target machine raspberrypi3, I also tried to compile raspberrypi3-64, which work ok for the YoeDistro itself, but doesn't work for YoeDistro/meta-wpe layer, where it fails on Cairo build. Should I create a new ticket for it?

timemaster5, 2020-03-16:
Dorry duplicated issue of https://github.com/WebPlatformForEmbedded/meta-wpe/issues/404

Closing

moorthy-bs, 2020-04-30:
fixed in #441

moorthy-bs, 2020-05-05:
@wouterlucas , this can be closed

closed 2020-05-05 by wouterlucas