WebPlatformForEmbedded/meta-wpe #422: vc4graphics and meta-wpe

limbouser, 2020-03-27:
MACHINE = "raspberrypi3"
MACHINE_FEATURES_append = " vc4graphics"
GPU_MEM_256 = "128"
GPU_MEM_512 = "196"
GPU_MEM_1024 = "396"
EXTRA_IMAGE_FEATURES = "debug-tweaks"
IMAGE_FEATURES_append = " ssh-server-dropbear hwcodecs"

DISTRO_FEATURES_remove += "x11"
DISTRO_FEATURES_append += " opengl"
DISTRO_FEATURES_append += " westeros"
When the local.conf was set as shown above, the built image didn't have the UI (ie, the hello display). I think the webkitbrowser doesn't work with with the above configuration. I must need vc4graphics for my application to work. Some of the drivers vc4_dri.so and a path /dev/dri/card0 etc gets activated only if I set vc4graphics. Is their any way to get around this?

wouterlucas, 2020-03-27:
Hi @limbouser I haven't crossed the vc4graphics / rpi4 bridge yet. Will probably spend some cycles on this in the future, but not a high priority for me right now.

I believe the `meta-webkit` guys have it running with `fdo`. Maybe that's worthwhile a shot if you can't wait.

limbouser, 2020-03-27:
Is their anyway I can use fdo as backend for meta-wpe? I'm new to all this. Can I use wpe and WebKit layers together?

wouterlucas, 2020-03-27:
Not unless you want to spend the effort of porting that over :)

But if you are just looking for WPEWebKit on an RPI https://github.com/Igalia/meta-webkit has all you need :)

limbouser, 2020-03-27:
Actually I wanted a wpeplugin setup too. For that meta-wpe is the solution right?

wouterlucas, 2020-03-27:
That is correct, I know @pwielders is doing some early development over at https://github.com/WebPlatformForEmbedded/buildroot but I don't think that's quite ready yet.

WPEFramework has no graphics dependencies, so that should work on pretty much anything. It is indeed WPEWebKit and its render path that needs to be properly setup.

limbouser, 2020-03-27:
Again, can someone try a combo of WebKit layer and WPE? Or running the WebKit in WPE with fdo backend? Are any of this doable? Considering I'm a noobie.

limbouser, 2020-04-13:
@wouterlucas If I can't have WPEFrameWork and vc4graphics, is this line in the readme section valid?
`Note: If you want to build a Wayland based image, don't remove Wayland and you can enable vc4graphics.`
I just wanna know if their is anyway I can work with both vc4graphics and WPE. :)

closed 2020-06-14 by limbouser