WebPlatformForEmbedded/meta-wpe #556: Error occured at building wpe-eglfs-image for raspberrypi3

suniltop, 2021-06-27:
Hi,

I have tried several times but no success yet.
I use master branch with dunfell branch of yoe. 
Because master brunch of yoe doesnt support main or master branch of meta-wpe. 

> Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-yoe-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "yoe"
DISTRO_VERSION       = "3.1.3"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
_meta-yoe             = "dunfell:7a2bf017cd28f0999c0114b4661ed9462a55380e"_
meta-96boards        = "HEAD:762e5e76ad1a5f385f845b09bde723edfcd09cba"
meta-altera          = "HEAD:5bb81b77d7bf81a726daec8819fa01c188863f5a"
meta-arm             
meta-arm-toolchain   = "HEAD:59974ccd5f1368b2a1c621ba3efd6d2c44c126dd"
meta-atmel           = "HEAD:32e9b0d4d320455d33999b8b9fb3b385cc4fc353"
meta-freescale       = "HEAD:12c3eed6a90b436565126d8ba6934f794b816e25"
meta-freescale-3rdparty = "HEAD:537cb85e4db17d1f6b356a92112b32ae2e39b235"
meta-odroid          
meta-odroid-extras   = "HEAD:f1f759209daa88ba52858a25b5ca13770d22e269"
meta-qcom            = "HEAD:1773ecb4285844a80109c84a0466d0531fca2da3"
meta-raspberrypi     = "HEAD:f0c75016f06c0395d1e76fde0ef1beb71eaf404a"
meta-riscv           = "HEAD:6f039004c2fa2be041c73d6ff711c86ae139faca"
meta-ti              = "HEAD:d54da744941ff97f2502bad2d7c51c5387bde929"
meta-clang           = "HEAD:e63d6f9abba5348e2183089d6ef5ea384d7ae8d8"
meta-qt5             = "HEAD:b4d24d70aca75791902df5cd59a4f4a54aa4a125"
meta-rust            = "HEAD:ab86e0a068e90c361b803f59a4baa79bf0c92ba7"
meta-elm-binary      = "HEAD:612d7b08ea064288d6297d10d4cf8c2fb3606379"
meta-filesystems     
meta-gnome           
meta-initramfs       
meta-multimedia      
meta-networking      
meta-oe              
meta-perl            
meta-python          
meta-webserver       
meta-xfce            = "HEAD:c38d2a74f762a792046f3d3c377827b08aade513"
meta                 = "HEAD:62ac245a4d6672bf8891530aab53d62cc3f6fbec"
meta-rockchip        = "HEAD:52e6a8c2c011cdc453649b0fac61e9b979033dc4"
_meta-wpe             = "master:5f16c3c7b7fbf2c1fa4e1cc9b275eabf93b00a78"_

Message:

```
`FAILED: lib/libWPEWebKit-0.1.so.2.2.1 
...
undefined reference to `JSC::Heap::vm() const'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.`
```

What should i do ?

TheLaGmAn, 2021-07-25:
I have the same issue, any hints?

suniltop, 2021-07-26:
I have solved recently myself.
You have to use this [patch](https://github.com/WebPlatformForEmbedded/meta-wpe/blob/main/recipes-wpe/wpewebkit/wpewebkit/0001-Fix-for-missing-heap-vm-main.patch)

Download patch file and save to .../yoe-distro/sources/meta-wpe/recipes-wpe/wpewebkit/wpewebkit/ folder
And then change param in wpewebkit file

```
SRC_URI = "git://github.com/WebPlatformForEmbedded/WPEWebKit.git;branch=wpe-2.22 \
           file://0001-WPEWebkit-compile-fix.patch \
"
```
with

```
SRC_URI = "\
    git://github.com/WebPlatformForEmbedded/WPEWebKit.git;branch=wpe-2.22 \
    file://0001-WPEWebkit-compile-fix.patch \
    file://0001-Fix-for-missing-heap-vm-main.patch \
"
```

Thus, patch will applied at compiling time.

If you will face any problem you should clean wpewebkit with commands that is belown and then will try compile again.

```
bitbake wpewebkit -ccleansstate
bitbake wpewebkit -cclean
```

TheLaGmAn, 2021-07-28:
Thanks for the patch! its almost there but im getting this error

ERROR: wpewebkit-2.22+gitAUTOINC+57bbecb942-r0 do_package_qa: QA Issue: wpewebkit rdepends on gstreamer1.0-plugins-bad-dash, but it isn't a build dependency? [build-deps]
ERROR: wpewebkit-2.22+gitAUTOINC+57bbecb942-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: Logfile of failure stored in: /.../yoe-dunfell/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpewebkit/2.22+gitAUTOINC+57bbecb942-r0/temp/log.do_package_qa.39244
ERROR: Task (/..../yoe-dunfell/sources/meta-wpe.git/recipes-wpe/wpewebkit/wpewebkit_2.22.bb:do_package_qa) failed with exit code '1'

im using dunfell branch, maybe its something on my local.conf? can you share yours or if  you have any other idea?

suniltop, 2021-07-28:
Hi

I guess that there is wrong something at sources/meta-wpe/recipes-wpe/wpewebkit/wpewebkit.inc
I given belown my own

```
SUMMARY = "WPE WebKit port pairs the WebKit engine with OpenGL-ES (OpenGL for Embedded Systems), \
           allowing embedders to create simple and performant systems based on Web platform technologies. \
           It is designed with hardware acceleration in mind, relying on EGL, the Wayland EGL platform, and OpenGL ES."
HOMEPAGE = "https://wpewebkit.org/"
LICENSE = "BSD & LGPLv2+"
LIC_FILES_CHKSUM = "file://Source/WebCore/LICENSE-LGPL-2.1;md5=a778a33ef338abbaf8b8a7c36b6eec80"

# WPE WebKit Common
DEPENDS += " \
    libwpe \
    bison-native ccache-native glib-2.0-native gperf-native libxml2-native ninja-native ruby-native chrpath-replacement-native \
    cairo freetype glib-2.0 gnutls harfbuzz icu jpeg pcre sqlite3 zlib \
    libepoxy libpng libsoup-2.4 libwebp libxml2 libxslt \
    virtual/egl virtual/libgles2 \
"

# unset CCACHE_DISABLE for this recipe
CCACHE_DISABLE[unexport] = "1"

S = "${WORKDIR}/git"

inherit cmake pkgconfig perlnative python3native

WPE_PLATFORM ?= "${@bb.utils.contains('DISTRO_FEATURES', 'wayland', 'westeros westeros-sink', 'egl', d)}"
WPE_PLATFORM ?= "${@bb.utils.contains('virtual/egl', 'broadcom-refsw', 'nexus', 'egl', d)}"
WPE_PLATFORM_x86 ?= "intelce"
WPE_PLATFORM ?= "egl"

PACKAGECONFIG ?= " \
    ${@bb.utils.contains('DISTRO_FEATURES', 'opencdm', 'opencdm', '', d)} \
    2dcanvas deviceorientation encryptedmedia fullscreenapi fetchapi gamepad indexeddb mediasource mediastatistics notifications nativevideo sampling-profiler subtitle subtlecrypto video webaudio ${WPE_PLATFORM} \
"

# Mesa only offscreen target support for Westeros backend
# FIXME Needs to be moved to mesa backend
PACKAGECONFIG[westeros-mesa] = "-DUSE_WPEWEBKIT_BACKEND_WESTEROS_MESA=ON,,"

# WPE Platform specific switches
PACKAGECONFIG[intelce] = "-DUSE_WPEWEBKIT_PLATFORM_INTEL_CE=ON -DUSE_HOLE_PUNCH_GSTREAMER=ON,,"
PACKAGECONFIG[nexus] = "-DUSE_WPEWEBKIT_PLATFORM_BCM_NEXUS=ON -DUSE_HOLE_PUNCH_GSTREAMER=ON,,"
PACKAGECONFIG[qcomdb] = "-DUSE_WPEWEBKIT_PLATFORM_QCOM_DB=ON,,"
PACKAGECONFIG[egl] = "-DUSE_GSTREAMER_GL=ON,-DUSE_GSTREAMER_GL=OFF,gstreamer1.0-plugins-base"
PACKAGECONFIG[westeros] = "-DUSE_WPEWEBKIT_PLATFORM_WESTEROS=ON,,westeros"

# westeros-sink, override with egl if glimagesink is vsink
PACKAGECONFIG[westeros-sink] = "-DUSE_HOLE_PUNCH_GSTREAMER=ON -DUSE_WESTEROS_SINK=ON -DUSE_GSTREAMER_HOLEPUNCH=ON,,westeros"
PACKAGECONFIG[wayland-sink]  = "-DUSE_WAYLAND_SINK=ON,,gstreamer1.0-plugins-bad"

# WPE features
PACKAGECONFIG[2dcanvas] = "-DENABLE_ACCELERATED_2D_CANVAS=ON,-DENABLE_ACCELERATED_2D_CANVAS=OFF,"
PACKAGECONFIG[deviceorientation] = "-DENABLE_DEVICE_ORIENTATION=ON,-DENABLE_DEVICE_ORIENTATION=OFF,"
PACKAGECONFIG[encryptedmedia] = "-DENABLE_ENCRYPTED_MEDIA=ON,-DENABLE_ENCRYPTED_MEDIA=OFF,libgcrypt"
PACKAGECONFIG[fetchapi] = "-DENABLE_FETCH_API=ON,-DENABLE_FETCH_API=OFF,"
PACKAGECONFIG[fullscreenapi] = "-DENABLE_FULLSCREEN_API=ON,-DENABLE_FULLSCREEN_API=OFF,"
PACKAGECONFIG[fusion] = "-DUSE_FUSION_API_GSTREAMER=ON,-DUSE_FUSION_API_GSTREAMER=OFF,"
PACKAGECONFIG[gamepad] = "-DENABLE_GAMEPAD=ON,-DENABLE_GAMEPAD=OFF,"
PACKAGECONFIG[geolocation] = "-DENABLE_GEOLOCATION=ON,-DENABLE_GEOLOCATION=OFF,geoclue"
PACKAGECONFIG[indexeddb] = "-DENABLE_INDEXED_DATABASE=ON,-DENABLE_INDEXED_DATABASE=OFF,"
PACKAGECONFIG[logs] = "-DENABLE_LOGS=ON,-DENABLE_LOGS=OFF,"
PACKAGECONFIG[mediasource] = "-DENABLE_MEDIA_SOURCE=ON,-DENABLE_MEDIA_SOURCE=OFF,gstreamer1.0 gstreamer1.0-plugins-good,${RDEPS_MEDIASOURCE}"
PACKAGECONFIG[mediastatistics] = "-DENABLE_MEDIA_STATISTICS=ON,-DENABLE_MEDIA_STATISTICS=OFF,"
PACKAGECONFIG[nativeaudio] = "-DENABLE_NATIVE_AUDIO=ON,-DENABLE_NATIVE_AUDIO=OFF,"
PACKAGECONFIG[nativevideo] = "-DENABLE_NATIVE_VIDEO=ON,-DENABLE_NATIVE_VIDEO=OFF,"
PACKAGECONFIG[notifications] = "-DENABLE_NOTIFICATIONS=ON,-DENABLE_NOTIFICATIONS=OFF,"
[... 104 lines cut ...]
    gstreamer1.0-omx \
    gstreamer1.0-plugins-bad-faad \
    gstreamer1.0-plugins-base-opengl \
"

# The RDEPS_EXTRA plugins are all required for certain media playback use cases,
# but have not yet been classified as being specific dependencies for video,
# webaudio or mediasource. Until that classification is done, add them all to
# each of the three groups...
RDEPS_MEDIASOURCE += "${RDEPS_EXTRA}"
RDEPS_VIDEO += "${RDEPS_EXTRA}"
RDEPS_WEBAUDIO += "${RDEPS_EXTRA}"

RDEPENDS_${PN} += " \
    virtual/libwpe \
    virtual/wpebackend \
"

RRECOMMENDS_${PN} += " \
    ca-certificates \
    ttf-bitstream-vera \
    shared-mime-info \
    ${PN}-web-inspector-plugin \
"
```

suniltop, 2021-09-28:
resolved

closed 2021-09-28 by suniltop