WebPlatformForEmbedded/meta-wpe #588: ERROR: Variable DISTRO_FEATURES_append contains an operation using the old override syntax

root-phantasoft, 2021-11-03:
The full error I get: 

yoe_add_layer git@github.com:WebPlatformForEmbedded/meta-wpe master
Adding existing repo at 'sources/meta-wpe' to the index
NOTE: Starting bitbake server...
NOTE: Started PRServer with DBfile: /root/yoe-distro/cache/prserv.sqlite3, Address: 127.0.0.1:36341, PID: 6082
ERROR: Variable DISTRO_FEATURES_append contains an operation using the old override syntax. Please convert this layer/metadata before attempting to use with a newer bitbake.
ERROR: Parse failure with the specified layer added, aborting.
please commit with - git add conf/bblayers.conf && git commit -s -m'Added module meta-wpe'

admindotnu, 2021-11-21:
use dunfell brache

primeroz, 2023-01-17:
I see this if a very old message , is the `dunfell` branch still the solution ?